swkim01/RPi.GPIO-PineA64 #3: fatal error: Python.h

valfa007, 2016-07-05:
when compile,there is an error:
building 'RPi._GPIO' extension
aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-aarch64-2.7/source/py_gpio.o
source/py_gpio.c:23:20: fatal error: Python.h: No such file or directory
 #include "Python.h"
                    ^
compilation terminated.
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

swkim01, 2016-07-06:
You have to install python-dev package before compiling.

SecT0uch, 2020-03-29:
Potentially solved and outdated. Can close this issue.

closed 2022-11-08 by swkim01