swkim01/RPi.GPIO-PineA64 #5: This module can only be run on a Raspberry Pi!

louwie17, 2016-08-15:
Hi, I am using Mate 1.12.1 on my Pine 64 and get the following when trying to run my python file:
Traceback (most recent call last):
  File "./blinking_led.py", line 4, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

This is after these steps:
sudo apt-get install python-pip
sudo pip install RPi.GPIO
sudo python setup.py install

The weird thing is this worked for me before, but I had to reinstall the system, and now I keep running in to the above issue.

louwie17, 2016-08-15:
I managed to get it working again, switching to python3 and skipping 'sudo pip install RPi.GPIO'
Installing it with: sudo python3 setup.py install

pklapperich, 2017-05-05:
What you experienced is expected behavior. When you did  'sudo pip install RPi.GPIO' you installed the original Rasberry Pi version of RPi.GPIO from pypi.python.org. To install the modified code that's been ported for the Pine64 you must manually install from this code repository.

exator, 2018-04-24:
I had this problem with python 2.7.
I found that I had that problem if I ran the python file inside RPi.GPIO-PineA64 folder.  As son as I  moved my .py file out of that folder and it worked a charm.

SecT0uch, 2020-03-29:
Solved. Can close this issue.

hellresistor, 2021-08-01:
i am getting this with python3 

```
$ python3 -V
Python 3.7.3
```

hellresistor, 2021-08-01:
problem solved with this way ;) 

```
 sudo apt-get purge python{,3}-rpi.gpio
 pip3 install RPi.GPIO
```

I think problem its when people install rpi.gpio  using `sudo apt-get install RPi.GPIO` command ;)

olexandr-mazepa, 2022-07-03:
It seems like ```/proc/cpuinfo``` changed the structure and doesn't contain expected info anymore.
Here is the output from  ```cat /proc/cpuinfo```:
```sh
pine64:~:% cat /proc/cpuinfo
processor	: 0
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

```
At the same time ```get_rpi_info``` function expects Hardware and Revision words
https://github.com/swkim01/RPi.GPIO-PineA64/blob/5ee3c2f1a4fb40fbe04fa8181cc8c9b49b445c6b/source/cpuinfo.c#L40-L61 
That's why the lib throws an exception here: https://github.com/swkim01/RPi.GPIO-PineA64/blob/5ee3c2f1a4fb40fbe04fa8181cc8c9b49b445c6b/source/py_gpio.c#L1023-L1026

This file may look differently only on my OS  because I use armbian but the Raspberrypi repo has a similar issue that may be useful https://github.com/raspberrypi/linux/issues/2110.

dest4, 2022-07-24:
On a Pine64, I bypassed the faulty detection the following way
```
diff --git a/source/cpuinfo.c b/source/cpuinfo.c
index 83d8379..9ec64f0 100644
--- a/source/cpuinfo.c
+++ b/source/cpuinfo.c
@@ -37,28 +37,29 @@ int get_rpi_info(rpi_info *info)
    int found = 0;
    int len;
 
-   if ((fp = fopen("/proc/cpuinfo", "r")) == NULL)
-      return -1;
-   while(!feof(fp)) {
-      fgets(buffer, sizeof(buffer), fp);
-      sscanf(buffer, "Hardware	: %s", hardware);
-      if (strcmp(hardware, "BCM2708") == 0 ||
-          strcmp(hardware, "BCM2709") == 0 ||
-          strcmp(hardware, "BCM2835") == 0 ||
-          strcmp(hardware, "BCM2836") == 0 ||
-          strcmp(hardware, "BCM2837") == 0 ) {
-         found = 1;
-         sscanf(buffer, "Revision	: %s", revision);
-      } else if (strcmp(hardware, "sun50iw1p1") == 0 ||
-                 strcmp(hardware, "Pine64") == 0 ||
-                 strcmp(hardware, "Pine64+") == 0 ) {
+//   if ((fp = fopen("/proc/cpuinfo", "r")) == NULL)
+//      return -1;
+//   while(!feof(fp)) {
+//      fgets(buffer, sizeof(buffer), fp);
+//      sscanf(buffer, "Hardware	: %s", hardware);
+//      if (strcmp(hardware, "BCM2708") == 0 ||
+//          strcmp(hardware, "BCM2709") == 0 ||
+//          strcmp(hardware, "BCM2835") == 0 ||
+//          strcmp(hardware, "BCM2836") == 0 ||
+//          strcmp(hardware, "BCM2837") == 0 ) {
+//         found = 1;
+//         sscanf(buffer, "Revision	: %s", revision);
+//      } else if (strcmp(hardware, "sun50iw1p1") == 0 ||
+//                 strcmp(hardware, "Pine64") == 0 ||
+//                 strcmp(hardware, "Pine64+") == 0 ) {
          pinea64_found = 1;
          found = 1;
+sprintf(hardware, "Pine64");
          sprintf(revision, "000b");
-      } else
-         sscanf(buffer, "Revision	: %s", revision);
-   }
-   fclose(fp);
+//      } else
+//         sscanf(buffer, "Revision	: %s", revision);
+//   }
+//   fclose(fp);

```

swkim01, 2022-07-30:
> On a Pine64, I bypassed the faulty detection the following way
>
> ```
> diff --git a/source/cpuinfo.c b/source/cpuinfo.c
> index 83d8379..9ec64f0 100644
> --- a/source/cpuinfo.c
> +++ b/source/cpuinfo.c
> @@ [...]

I committed your code just now. Thank you.

dest4, 2022-07-30:
Beware, IIRC there were different code paths for Pine64 and Pine64+

closed 2022-11-08 by swkim01