swkim01/RPi.GPIO-PineA64 #9: python3 ImportError: dynamic module does not define module export function (PyInit__SPI)

dolfandringa, 2017-06-16:
After compiling for python3 I get the following error when trying to import the module:

```
>>> import RPi.SPI as spi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/RPi/SPI/__init__.py", line 4, in <module>
    from RPi._SPI import *
ImportError: dynamic module does not define module export function (PyInit__SPI)
```

With python2 this doesn't happen (the module imports just fine, but haven't tested anything else). Should python3 work as well? The app in which I would like to use it is built on python3. Would it be hard to add python3 support?

Cheers,

Dolf.

dolfandringa, 2017-06-16:
I made a fix with PR #10

SecT0uch, 2020-03-29:
Solved, can be closed.

closed 2022-11-08 by swkim01