swkim01/RPi.GPIO-PineA64 #14: Can't install because of  "error: expected ‘(’ before ‘PyType_HasFeature’"

pacsinta, 2022-06-16:
I installed python3 and python3-dev with apt install.
Then I typed: sudo python3 setup.py install

I've got a lot of warnings, but some errors too.

<br>
The error messages are:
 
`/usr/include/python3.9/object.h:633:41: error: expected ‘(’ before ‘PyType_HasFeature’
  633 | #define PyType_FastSubclass(type, flag) PyType_HasFeature(type, flag)`

and:

`source/py_gpio.c:247:37: error: expected ‘;’ before ‘{’ token
  247 |    } else if PyList_Check(chanlist) {`

uomodellaforesta, 2022-11-03:
Hello, i've the same issue! have you solved it?

pacsinta, 2022-11-03:
> Hello, i've the same issue! have you solved it?

Unfortunately no 😢

uomodellaforesta, 2022-11-04:
Hi pacsinta, i found two errors into the file py_gpio.c but now i found a new issue.
However i corrected two "else if" statements. At the line 247 (as you see in your logs) and 249 there is not any round brackets!
i added the brackets but unfortunately not work for me

uomodellaforesta, 2022-11-04:
i tried to install the original RPi.GPIO module for raspberry into pine64. I seen that work only the 0.7.1 version for me and the previous versions makes some errors like the newest seen for pine64.

uomodellaforesta, 2022-11-04:
Hi pacsinta, i seen changelogs for 0.7.1 version and i seen some information about "Fix build using GCC 10 (issue 187)".
This issue is other thing but i installed the package with the command: 

CFLAGS="-fcommon" pip install -e RPi.GPIO-PineA64-master/
PS: RPi.GPIO-PineA64-master/ is my folder where setup.py is located.

Now i need to test it! i hope that this is useful for you

Marco

uomodellaforesta, 2022-11-05:
Hello pacsinta! i tested and it works :)